RadixIoT/BACnet4J #1: Implement SO_REUSEPORT

terrypacker, 2016-02-11:
From a comment on this commit cfd52615fa3bfb93ce8150ecbaa8ec81323960c9 

How about reusePort to (hopefully) allow multiple instances to use the same interface?

SO_REUSEPORT is what most people would expect SO_REUSEADDR to be. Basically, SO_REUSEPORT allows you to bind an arbitrary number of sockets to exactly the same source address and port as long as all prior bound sockets also had SO_REUSEPORT set before they were bound.

[More info here](http://stackoverflow.com/questions/14388706/socket-options-so-reuseaddr-and-so-reuseport-how-do-they-differ-do-they-mean-t/14388707#14388707)

mlohbihler, 2017-02-15:
Java only has setReuseAddress AFAICS. What is the proposal here? To change the name of the field/method?

Frozenlock, 2017-02-22:
Looks like you are right... for now.
SO_REUSEPORT [should be present in Java 9](http://hg.openjdk.java.net/jdk9/dev/jdk/rev/f92af283ab18).

Frozenlock, 2017-09-29:
Now that Java 9 is out, I'd like to bring back the attention on this issue. :-)

(Hopefully this simple change would allow multiple instances of BACnet4J to run on a single interface/IP)

Frozenlock, 2019-07-18:
This appears to have been added, so this issue can probably be closed.

A side question : is it possible to have multiple Bacnet devices using the same IP and communicate normally between them? I've given it a quick try, but all the devices on the same IP are answering the readProperty requests at the same time, even if they are the one that sent the request..

terrypacker, 2019-10-11:
@Frozenlock I'm going to close this but I wrote a test awhile back so here is a starting point if you want:
```

closed 2019-10-11 by terrypacker