RadixIoT/BACnet4J #2: add a way to update the remote devices list

Frozenlock, 2016-02-29:
Some devices on a BACnet network are only connected temporarily (think of a laptop from a maintenance guy). If BACnet4J doesn't ping the remote devices periodically to determine if they are still there, it should at least be possible to manually remove them from the remote devices list.

I would suggest adding a `removeRemoteDevice` method in the [remote devices management](https://github.com/infiniteautomation/BACnet4J/blob/master/src/main/java/com/serotonin/bacnet4j/LocalDevice.java#L448) section.

Frozenlock, 2016-05-14:
Any news on this?

mlohbihler, 2017-02-15:
v4 has caches for remote devices, objects, and properties, and includes configurable expiration policies. See LocalDevice.getCachePolicies

Frozenlock, 2017-02-22:
Oh, derp, it's on the V4 branch.

closed 2017-02-22 by Frozenlock