RadixIoT/BACnet4J #3: endless loop in transport log/error message?

Frozenlock, 2016-02-29:
Whenever I get a BACnetTimeoutException, it *endlessly* prints this message in the console:

```
[BACnet4J transport] ERROR com.serotonin.bacnet4j.transport.DefaultTransport - Error during expire messages: 
com.serotonin.bacnet4j.exception.BACnetTimeoutException
	at com.serotonin.bacnet4j.transport.DefaultTransport.expire(DefaultTransport.java:810)
	at com.serotonin.bacnet4j.transport.DefaultTransport.run(DefaultTransport.java:414)
	at java.lang.Thread.run(Thread.java:745)
```

(You can easily generate a timeout exception by trying to read a property of a device no longer present on the network.)

Frozenlock, 2016-02-29:
I just realized that version 3.2.4 is out. I don't get this message anymore.

Frozenlock, 2016-03-01:
Apologies, I was too quick to close this one. Even with 3.2.4 I still get the endless printing of the error message. 

It starts approximately 19 seconds after making a request to read the properties of a device no longer on the network.

The APDU related configurations of the local device are as follow : 

```
apdu-timeout : 6000
apdu-segment-timeout : 5000
number-of-apdu-retries : 2
```

mlohbihler, 2017-02-15:
I created a test case to try and recreate this, but was not successful. See https://github.com/infiniteautomation/BACnet4J/commit/0b9a5e63c4c2222d79e2d3593b2266d099b6d54c. Can you try this out yourself? One version that i used had a sleep at the end to see if any messages started appearing.

closed 2019-07-18 by Frozenlock