RadixIoT/BACnet4J #4: Network Routing is not unique enough

gada, 2016-03-07:
Hi 
I would like to bring to your attention an issue (I'm not sure if its a bug or not yet) we have found with the latest BACNET 4j stack (3.2.4).
We have network configuration in one of the buildings that has multiple BACNET networks with same id and same Network router MAC address.
So for example:
Remote Device 1 , MSTP station ID=2    with Network ID:2 and Linked service x.x.x.x
Remote Device 2 , MSTP station ID=2    with Network ID:2 and Linked service y.y.y.y
 
In that case the BACNET STACK cannot route properly since DefaultTransport is doing the routing using network number only.
 
I have opened an issue at the forum: http://infiniteautomation.com/forum/topic/2141/network-routers-map-in-default-transport-is-not-unique 

I have a temporary resolution on gist 
https://gist.github.com/gada/a43d98d2c10fe62fa880

Basically what I did is try to route by remote device ID and if I don’t have the routing per device information I will try to route by network number  as you originally do.

mlohbihler, 2017-02-15:
I didn't think it was valid to have multiple networks with the same id.

gada, 2017-02-22:
Unfortunately we have few sites that are have multiple networks with the same ID.

Are you intend to upgrade the stack to support that ?

mlohbihler, 2017-02-22:
But that was kind of the point: just because you have networks with the same id doesn't make it valid. You should probably change your networks.

gada, 2017-02-22:
The problem that those networks are not in our reach - We have no way of telling other buildings how to set their networks even if they are wrong.

From practical point of view we have an issue here.

I have already change the stack implementation to deal with that but was expecting to have the solution by you as now it will be hard for us to merge the code.

cneilson, 2017-02-22:
There must not be multiple networks in a single installation with the same network number. Many products will fail to work if this is the case.

It can also be shown that it will be impossible to communicate with any device on a network with a duplicate number if there are 2 routers between the two device that are to communicate with each other.

So instead of trying to "fix" the BACnet4J stack, get the problem sites fixed.

mlohbihler, 2017-02-23:
Some additional comments from Clifford Copass:

> Some notes from 135-2012 to build on Carl’s comments:
> Bottom of clause 5.1
> “A BACnet Device is uniquely located by an NSAP, which consists of a network number and a MAC address.”
>
> First paragraph in clause 6
> “A BACnet Device is uniquely located by a network number and a MAC address.”
>
> Clause 6.6.3.2
> “If the network number is not found in the routing table, the router shall attempt to discover the next router on the path to the indicated destination network by generating a Who-Is-Router-To-Network message containing the specified destination network number and broadcasting it out all its ports other than the one from which the Who-Is-Router-To-Network message arrived.”
>
> Comments:
> MAC addresses are commonly duplicated, MS/TP being a prime example, meaning the network numbers must be unique.
>If a router gets two responses to a broadcast Who-Is-Router-To-Network message, how does it know which one is correct?  The answer is that it can’t and this configuration is not supported by BACnet.

mlohbihler, 2017-02-23:
Closing this issue.

closed 2017-02-23 by mlohbihler

gada, 2017-02-23:
Hi Guys.

I appreciate your response and like I have already said there is no way for me as a client of many buildings to change the way they are set up.

Beside that - few comments regarding your response:

First: It took you almost one year to get back to my call - Way too long - You at least could apologise.

Second: You have made a fundamental change in the stack without notifying your clients about the potential risk and that’s is bad.

Third: From design point of view I think the stack should not make smart decision or assumptions about the network as most of them today as you might know are not architecture like the book.

Four: In many cases I have seen that "IAM router to Network" is not being submitted by BACNET routers so the network address is not known by the stack and relying on that is not working in most cases. The way is used to work where it got the network routers from IAM response PDU is much more stable – I still don’t see why removing that information from the Object.

Five: What do you call installation? - If my stack is accessing few sites behind BBMD with potential identical network numbers is the driver build to deal with that?

Regards.

mlohbihler, 2017-02-23:
1) Sorry about that, but note that i am not the owner of the stack, just a regular contributor. I receive no financial or other benefit from addressing these issues. I just happened to see them, and act on them, when i did. Of course, if you want to pay for support, that can be arranged.

2) Making breaking changes to software is fairly common to do, depending on the package, and this is usually signaled by increasing the major version number. More can be done to describing the breaks in the documentation; we haven't generally done a good job at that. Note that version 4 makes several breaks with version 3. I would notify everyone that this is the case, but honestly i don't know who the users are: people don't usually tell me. If you bought a license from me (while i still owned the software), then i'm afraid i don't know who you are. If you bought it from IA, you should have a chat with them.

As for your other points, if you can demonstrate where the software is doing something non-compliant, that would be much appreciated. Suggestions for improvement are also welcome, and may actually get worked on if someone has time, but it helps to be specific about what it does now and what you want it to do.

Matthew