RadixIoT/BACnet4J #7: Out of order segmented messages not reassembled correctly

d-shapiro, 2016-06-03:
I noticed this when trying to send a ReadRange request and read the response.
In this particular case, the response came in as a segmented Complex-ACK, and the segments arrived in the order: 0, 1, 2, 5, 3, 4. However, the resulting ByteQueue was missing all of the bytes from segments 3 and 4, causing a "missing required parameter" error when trying to decode the incomplete ByteQueue. 
It looks like DefaultTransport assembles segments in the order they arrive, and stops assembling once a segment arrives with moreFollows = false.

mlohbihler, 2017-02-15:
Hmm, interesting case. I will take a look. Does this happen often? I've never seen this before.

mlohbihler, 2017-02-16:
Turns out there were a few problems with segments being received out of order.  I committed a fix for this here: https://github.com/infiniteautomation/BACnet4J/commit/381b6f67e8c853b262f0411330a652d318c89e99