RadixIoT/BACnet4J #13: TimerTask can not work with RequestUtils

char1st, 2017-09-14:
I want run a periodic task  use Timertask and ScheduledExecutorService  to run  RequestUtils.getProperties , but is not work , and I found there is no request send to remoteDevice. 
```
	static void find(Long period, TimeUnit unit, Boolean isTest) {
		try {
			SequenceOf<ObjectIdentifier> objList = RequestUtils.getObjectList(localDevice, d);
			List<ObjectIdentifier> logOids = getLogOids(objList);
			TimerTask task = new TimerTask() {
				@Override
				public void run() {
					for (ObjectIdentifier i : logOids) {
						try {
							List<ReadAccessSpecification> specs = new ArrayList<ReadAccessSpecification>();
							specs.add(new ReadAccessSpecification(i, PropertyIdentifier.presentValue));
							specs.add(new ReadAccessSpecification(i, PropertyIdentifier.units));
							specs.add(new ReadAccessSpecification(i, PropertyIdentifier.objectName));
							specs.add(new ReadAccessSpecification(i, PropertyIdentifier.description));
							specs.add(new ReadAccessSpecification(i, PropertyIdentifier.objectType));
							ReadPropertyMultipleRequest multipleRequest = new ReadPropertyMultipleRequest(
									new SequenceOf<ReadAccessSpecification>(specs));
							localDevice.send(d, multipleRequest);
							ReadPropertyMultipleAck send = localDevice.send(d, multipleRequest).get();
							SequenceOf<ReadAccessResult> readAccessResults = send.getListOfReadAccessResults();
							System.out.print(i.getInstanceNumber() + " " + i.getObjectType() + ", ");
							for (ReadAccessResult result : readAccessResults) {
								for (ReadAccessResult.Result r : result.getListOfResults()) {
									System.out.print(r.getReadResult() + ", ");
								}
							}
						} catch (Exception e) {
						}
					}
				}
			};
			ScheduledExecutorService scheduledPool = Executors.newSingleThreadScheduledExecutor();
			scheduledPool.scheduleWithFixedDelay(task, 0, period, unit);
		} catch (Exception e) {
			e.printStackTrace();
		}
	}
```
What is the rigth way use  RequestUtils in Timertask ?

mlohbihler, 2017-09-14:
This is not the problem, but you are using a TimerTask only as a Runnable. You would normally use it with a java.util.Timer. Also, the local device already has a scheduled executor, so you can simplify the async code by doing this:

```
        localDevice.scheduleWithFixedDelay(() -> {
            for (final ObjectIdentifier i : logOids) {
                ...
            }
        }, 0, period, unit);
```

But back to your question. What is the problem you are seeing when you run async?

char1st, 2017-09-15:
THX  . Please excuse me for my poor English. 
     My problem  is  When  I use the first  method , the code lookes like  hang 
at localDevice.send(d, multipleRequest) ,and there is no bacnet package send to remoteDevice  show in the wireshark .

mlohbihler, 2017-09-15:
It could be that while your async code was running the local device was terminated in the other thread.

char1st, 2017-09-18:
I found the local device in  the first method is not isInitialized , and had to initialize agine then  terminate
it in the end.
I will use the  localDevice.scheduleWithFixedDelay() .

closed 2017-09-18 by char1st

mlohbihler, 2017-09-18:
I have added a check to ensure that the local device is initialized before it will attempt to send requests.