RadixIoT/BACnet4J #14: Broken dependency

egcodes, 2017-11-27:
For these libraries maven is giving 404 error.

Could not resolve dependencies for project com.serotonin:bacnet4j:jar:4.1.2: Failed to collect dependencies at lohbihler:sero-scheduler:jar:[1.0.0,2): No versions available for lohbihler:sero-scheduler:jar:[1.0.0,2) within specified range -> [Help 1]

``` 
        <dependency>
            <groupId>lohbihler</groupId>
            <artifactId>sero-scheduler</artifactId>
            <version>[1.0.0,2)</version>
        </dependency>

        <dependency>
            <groupId>lohbihler</groupId>
            <artifactId>sero-warp</artifactId>
            <version>[1.0.0,2)</version>
        </dependency>
```

splatch, 2017-11-27:
> For these libraries maven is giving 404 error.

Make sure you included in your project additional repository where these binaries are located. They are not pushed to maven central so you need to point where they are.

Chunk of config for project or your maven profile.
```xml
    <repositories>
        <repository>
            <releases>
                <enabled>false</enabled>
            </releases>
            <snapshots>
                <enabled>true</enabled>
            </snapshots>
            <id>ias-snapshots</id>
            <name>Infinite Automation Snapshot Repository</name>
            <url>https://maven.mangoautomation.net/repository/ias-snapshot/</url>
        </repository>
        <repository>
            <releases>
                <enabled>true</enabled>
            </releases>
            <snapshots>
                <enabled>false</enabled>
            </snapshots>
            <id>ias-releases</id>
            <name>Infinite Automation Release Repository</name>
            <url>https://maven.mangoautomation.net/repository/ias-release/</url>
        </repository>
    </repositories>
```

egcodes, 2017-11-27:
Another error
```
Downloading: https://maven.mangoautomation.net/repository/ias-release/org/mockito/mockito-core/maven-metadata.xml
Downloading: https://maven.mangoautomation.net/repository/ias-snapshot/org/mockito/mockito-core/maven-metadata.xml

Could not transfer metadata org.mockito:mockito-core/maven-metadata.xml from/to ias-snapshots (https://maven.mangoautomation.net/repository/ias-snapshot/): sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: timestamp check failed
Could not transfer metadata org.mockito:mockito-core/maven-metadata.xml from/to ias-releases (https://maven.mangoautomation.net/repository/ias-release/): sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: timestamp check failed

```

splatch, 2017-11-27:
Ok, all clear - security certificate expired 4 days ago. Definitely something which needs to be solved on other side.

closed 2017-11-27 by egcodes

terrypacker, 2017-11-27:
Yes sorry about that.  We will get the server back up and running as soon as possible.

AlexMAS, 2019-11-29:
Hello,

Probably we have this bug again...

```
   > Could not resolve lohbihler:sero-scheduler:[1.0.0,2).
     Required by:
         project :com.energymachines.scada.node > project :com.energymachines.scada.network > project :com.energymachines.scada.connector.bacnet > com.serotonin:bacnet4j:4.1.6
      > Failed to list versions for lohbihler:sero-scheduler.
         > Unable to load Maven meta-data from https://maven.mangoautomation.net/repository/ias-release/lohbihler/sero-scheduler/maven-metadata.xml.
            > Could not HEAD 'https://maven.mangoautomation.net/repository/ias-release/lohbihler/sero-scheduler/maven-metadata.xml'.
               > sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed
   > Could not resolve lohbihler:sero-warp:[1.0.0,2).
     Required by:
         project :com.energymachines.scada.node > project :com.energymachines.scada.network > project :com.energymachines.scada.connector.bacnet > com.serotonin:bacnet4j:4.1.6
      > Failed to list versions for lohbihler:sero-warp.
         > Unable to load Maven meta-data from https://maven.mangoautomation.net/repository/ias-release/lohbihler/sero-warp/maven-metadata.xml.
            > Could not HEAD 'https://maven.mangoautomation.net/repository/ias-release/lohbihler/sero-warp/maven-metadata.xml'.
               > sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed
```

P.s. Are these dependencies really needed at runtime? Or they are used only for testing?

terrypacker, 2019-12-02:
Our maven repository went down over the weekend.  As for the P.S.  yes they are used at runtime, I reviewed that last week so we cannot make them test only.