RadixIoT/BACnet4J #18: when i tried to serizalize an Array of RemoteDevice Object , I find i can't, because it referenced a  Address Object.  And the Address.class can't support serializable.   I  need your help~

duyanhan1995, 2018-03-26:
`java.io.WriteAbortedException: writing aborted; java.io.NotSerializableException: com.serotonin.bacnet4j.type.constructed.Address
	at java.io.ObjectInputStream.readObject0(Unknown Source)
	at java.io.ObjectInputStream.defaultReadFields(Unknown Source)
	at java.io.ObjectInputStream.readSerialData(Unknown Source)
	at java.io.ObjectInputStream.readOrdinaryObject(Unknown Source)
	at java.io.ObjectInputStream.readObject0(Unknown Source)
	at java.io.ObjectInputStream.readObject(Unknown Source)
	at java.util.ArrayList.readObject(Unknown Source)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at java.io.ObjectStreamClass.invokeReadObject(Unknown Source)
	at java.io.ObjectInputStream.readSerialData(Unknown Source)
	at java.io.ObjectInputStream.readOrdinaryObject(Unknown Source)
	at java.io.ObjectInputStream.readObject0(Unknown Source)
	at java.io.ObjectInputStream.defaultReadFields(Unknown Source)
	at java.io.ObjectInputStream.readSerialData(Unknown Source)
	at java.io.ObjectInputStream.readOrdinaryObject(Unknown Source)
	at java.io.ObjectInputStream.readObject0(Unknown Source)
	at java.io.ObjectInputStream.readObject(Unknown Source)
	at com.cn.yisquare.bacnet.getpointinfo.LaterScanToJson.init(LaterScanToJson.java:67)
	at com.cn.yisquare.bacnet.util.PowerfulDecoration.init(PowerfulDecoration.java:28)
	at com.cn.yisquare.bacnet.getpointinfo.Client.getAndUploadTask(Client.java:84)
	at com.cn.yisquare.bacnet.getpointinfo.Client.access$0(Client.java:81)
	at com.cn.yisquare.bacnet.getpointinfo.Client$1.run(Client.java:63)
	at java.util.TimerThread.mainLoop(Unknown Source)
	at java.util.TimerThread.run(Unknown Source)
Caused by: java.io.NotSerializableException: com.serotonin.bacnet4j.type.constructed.Address
	at java.io.ObjectOutputStream.writeObject0(Unknown Source)
	at java.io.ObjectOutputStream.defaultWriteFields(Unknown Source)
	at java.io.ObjectOutputStream.writeSerialData(Unknown Source)
	at java.io.ObjectOutputStream.writeOrdinaryObject(Unknown Source)
	at java.io.ObjectOutputStream.writeObject0(Unknown Source)
	at java.io.ObjectOutputStream.writeObject(Unknown Source)
	at java.util.ArrayList.writeObject(Unknown Source)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at java.io.ObjectStreamClass.invokeWriteObject(Unknown Source)
	at java.io.ObjectOutputStream.writeSerialData(Unknown Source)
	at java.io.ObjectOutputStream.writeOrdinaryObject(Unknown Source)
	at java.io.ObjectOutputStream.writeObject0(Unknown Source)
	at java.io.ObjectOutputStream.defaultWriteFields(Unknown Source)
	at java.io.ObjectOutputStream.writeSerialData(Unknown Source)
	at java.io.ObjectOutputStream.writeOrdinaryObject(Unknown Source)
	at java.io.ObjectOutputStream.writeObject0(Unknown Source)
	at java.io.ObjectOutputStream.writeObject(Unknown Source)`

terrypacker, 2018-03-27:
@duyanhan1995 Address extends BaseType which extends Encodable.  RemoteDevice probably should not implement Serializable as most of its fields are not Serializable.  Feel free to create a pull request that implements Serializable for the required fields on RemoteDevice.

duyanhan1995, 2018-03-28:
@terrypacker How do i understand "create a pull request "? I think i am awkword.  There also exist a "userData" field which type is "Object"  in RemoteDevice, and we can't serizalize it. Could you give some code hints?

closed 2018-09-10 by duyanhan1995