RadixIoT/BACnet4J #22: Insecure HTTP dependencies

WhittlesJr, 2018-05-21:
Despite the fix for #20 that went into 4.1.5, I am still having "Insecure HTTP repo" problems. 

I've identified the specific sources:

```
HTTP Wagon get: lohbihler/sero-scheduler/maven-metadata.xml from #object[java.io.File 0x4de273ae /home/alexj/.m2/repository/lohbihler/sero-scheduler/maven-metadata-jcenter.xml.part]
HTTP Wagon get: org/slf4j/slf4j-api/maven-metadata.xml from #object[java.io.File 0x62e1d0db /home/alexj/.m2/repository/org/slf4j/slf4j-api/maven-metadata-jcenter.xml.part]
HTTP Wagon get: org/slf4j/slf4j-api/maven-metadata.xml.sha1 from #object[java.io.File 0x411052bc /home/alexj/.m2/repository/org/slf4j/slf4j-api/maven-metadata-jcenter.xml.sha1-260f3ac51103864788133500845.tmp]
HTTP Wagon get: lohbihler/sero-warp/maven-metadata.xml from #object[java.io.File 0x11f8d7b /home/alexj/.m2/repository/lohbihler/sero-warp/maven-metadata-jcenter.xml.part]
```

I'm frankly not sure where a fix would need to be made...

WhittlesJr, 2018-06-20:
Does anybody have any thoughts on this?

splatch, 2018-06-20:
Hey @WhittlesJr,
As bacnet4j [contains "jcenter" repository](https://github.com/infiniteautomation/BACnet4J/blob/8eab024fbcd48e2586718f2a4651dc67812e3bb9/pom.xml#L83), probably for some of it's dependencies, you keep getting metadata from it. In order to avoid scanning of jcenter all bacnet4j deps have to be deployed to maven central. Then extra repos will be unecessary in pom.xml.

If you are afraid of http connection jcenter should already be available via https.

Cheers,
Lukasz

WhittlesJr, 2018-06-20:
Hi! It's not so much that I'm afraid of HTTP, but that it breaks my build ([Leiningen](https://leiningen.org/)). I currently have to put in hacks to build either [bacure](https://github.com/Frozenlock/bacure) or my own project (which depends on bacure). 

What confuses me is that I see that 4.1.5 of BACnet4J switched to using HTTPS for the jcenter repo, yet I still experience this issue. So is it really still jcenter that is the problem? Where exactly does a fix need to go?

terrypacker, 2018-06-20:
Are you building via Maven?  If so it could be that we missed adding https to one of the jcenter repos:

https://github.com/infiniteautomation/BACnet4J/blob/8eab024fbcd48e2586718f2a4651dc67812e3bb9/pom.xml#L83

WhittlesJr, 2018-06-20:
Ah, I bet that's it!

WhittlesJr, 2018-06-27:
Thank you for resolving this. Will this end up in a near-term bugfix release?

terrypacker, 2018-06-27:
Built and released as 4.1.6.

closed 2018-06-27 by terrypacker