RadixIoT/BACnet4J #23: v3 to v4 : Non-IP devices no longer discovered?

Frozenlock, 2018-08-03:
My network has one BACnet/IP device able to bridge to the rest of the MS/TP network.

In v3 I was able to see and communicate with every devices (IP and MS/TP) via BACnet/IP.

Since V4 the only devices I can see/communicate with are the BACnet/IP ones.
Is there a new configuration or listener I need to use to be able to communicate with the MS/TP devices?

JCP17, 2018-11-13:
I have the same problem but at migration from V2 to 3.2.2.
Is there any information on this yet?

Frozenlock, 2018-11-14:
No. I've narrowed down the problem to the handle (https://github.com/infiniteautomation/BACnet4J/blob/master/src/main/java/com/serotonin/bacnet4j/service/unconfirmed/IAmRequest.java#L84).

It appears that if something goes wrong there, the remote device isn't added to the list and the `fireIAmReceived` is never fired.

Frozenlock, 2019-07-18:
[V5 : I confirm that the problem is coming from `DiscoveryUtils.getExtendedDeviceInformation`. When it fails, the extended information is never fetched again and the `IAmReceived` event is never fired.

I fixed it in a fork by removing the call to `DiscoveryUtils.getExtendedDeviceInformation` from the `IAmRequest` handling and adding a similar function to a `IAmReceived` listener. This way the extended info retrieval is entirely configurable and doesn't screw up the event when encountering an error.

terrypacker, 2019-07-19:
Any chance you can make a PR for that fix from your Fork?

Thanks!

lovefamilyC, 2019-11-20:
Brother, according to this method can indeed get the remote device object, but always return to the object property when reporting BACnetTimeoutException

PR #40 (2019-11-22): Do not get the device extended info when handling the IAm request.
This fixes #23 

However be warned that this is a breaking change for any user of this library : the function to get the extended info must be manually added via a `IAmRequest` listener.

While this requires a little more setup, I prefer this approach as it leaves the process entirely configurable. (I don't use `DiscoveryUtils.getExtendedDeviceInformation` at all, so if it's possible to not hardcode it anywhere I would really appreciate.)
closed 2019-11-22 by terrypacker