RadixIoT/BACnet4J #25: can I make local device in another ip?

yzkim9501, 2018-09-11:
```
 IpNetworkBuilder inb = new IpNetworkBuilder().port(47808);       
 IpNetwork network = inb.build();
 Transport transport = new DefaultTransport(network);
 localDevice = new LocalDevice(Integer.parseInt(properties.getProperty("deviceId")), transport);
 localDevice.initialize();
 localDevice.getEventHandler().addListener(new Listener());
 localDevice.sendGlobalBroadcast(new WhoIsRequest());
 LOGGER.info("Initialized");
```

I want to create local device in another ip not localhost, like "192.168.1.200"
Can I create it by modifying this code?

Or, If I want to make 'client bacnet4j object' and 'server bacnet4j object' in same ip, like In process A, I broadcast bacnet through bacnet4j, and In process B, I recieve brodcasting bacnet through bacnet4j. I am getting bind error because of same port(47808) and same ip(localhost). Can I get it? 
Is there any test code of it? I tried to find it, but I couldn't from the sample test code. thank you for reading. :)

yzkim9501, 2018-09-11:
Let me specify the issue. 
I have monitoring process(let me call it A) that recieving bacnet devices using bacnet4j. It works through localDevice, listener. recieving 192.168.x.x bacnet remote devices.
I have to make a bacnet local device(let me call it B) to broadcast virtual bacnet device, but there are bind exception because of monitoring process already using local device, local host, port num 47808.
I tried one of process(A,B) of switching port number to another(47809), It works, but A cannnot monitor B because of different port number(ofcourse).

lovefamilyC, 2019-11-19:
> 让我指定问题。
> 我有一个监视过程（我称之为A），它使用bacnet4j接收bacnet设备。它通过侦听器localDevice起作用。接收192.168.xx bacnet远程设备。
> 我必须制作一个bacnet本地设备（我称其为B）来广播虚拟bacnet设备，但是由于已经使用本地设备，本地主机，端口号47808监视进程，因此存在绑定异常。

> Let me specify the issue.
> I have monitoring process(let me call it A) that recieving bacnet devices using bacnet4j. It works through localDevice, listener. recieving 192.168.x.x bacnet remote [...]

Brother, is your problem solved? I also encountered the same problem, the local device cannot scan to the VTS device because it is different from the port number of the VTS device.

terrypacker, 2019-11-19:
This code will bind to the first usable BACnet interface on port 9000 with 2 separate local devices on the same network.
```
        List<InterfaceAddress> usable = BacnetIpUtils.listUsableBACnetInterfaces();
        Assume.assumeTrue(usable.size() > 0);
        
        InterfaceAddress address = usable.get(0);
        String bindAddress = address.getAddress().toString().split("/")[1];
        String broadcastAddress = address.getBroadcast().toString().split("/")[1];
        
        //Configure the first network, ensure we set reuse address
        IpNetwork networkOne = new IpNetworkBuilder()
                .withLocalBindAddress(bindAddress)
                .withBroadcast(broadcastAddress, address.getNetworkPrefixLength())
                .withLocalNetworkNumber(1).withPort(9000).withReuseAddress(true).build();
        Transport transportOne = new DefaultTransport(networkOne);
        LocalDevice localDeviceOne = new LocalDevice(1, transportOne);
        
        
        
        IpNetwork networkTwo = new IpNetworkBuilder()
                .withLocalBindAddress(bindAddress)
                .withBroadcast(broadcastAddress, address.getNetworkPrefixLength())
                .withLocalNetworkNumber(1).withPort(9000).withReuseAddress(true).build();
        Transport transportTwo = new DefaultTransport(networkTwo);
        LocalDevice localDeviceTwo = new LocalDevice(2, transportTwo);
        
        localDeviceOne.initialize();
        localDeviceTwo.initialize();
```

lovefamilyC, 2019-11-20:
> This code will bind to the first usable BACnet interface on port 9000 with 2 separate local devices on the same network.
>
> ```
>         List<InterfaceAddress> usable = [...]

Brother, thank you, your code has helped a lot!