RadixIoT/BACnet4J #26: Caused by: com.serotonin.bacnet4j.exception.BACnetRuntimeException: No enumerated found for name 'analogInput'

duyanhan1995, 2018-10-19:
There is a method "forName(String):ObjectType" in the ObjectType Class.But when I try to get ObjectType by name, it's go wrong :

[image]

duyanhan1995, 2018-10-19:
Caused by: com.serotonin.bacnet4j.exception.BACnetRuntimeException: No enumerated found for name 'analogInput'
	at com.serotonin.bacnet4j.type.primitive.Enumerated.forName(Enumerated.java:95)
	at com.serotonin.bacnet4j.type.enumerated.ObjectType.forName(ObjectType.java:120)
	at com.yisquare.ba.scanner.config.PointConfig.getPointsAndProperties(PointConfig.java:47)
	at com.yisquare.ba.scanner.config.PointConfig$$EnhancerBySpringCGLIB$$c915987d.CGLIB$getPointsAndProperties$0(<generated>)
	at com.yisquare.ba.scanner.config.PointConfig$$EnhancerBySpringCGLIB$$c915987d$$FastClassBySpringCGLIB$$5b91cc6e.invoke(<generated>)
	at org.springframework.cglib.proxy.MethodProxy.invokeSuper(MethodProxy.java:228)
	at org.springframework.context.annotation.ConfigurationClassEnhancer$BeanMethodInterceptor.intercept(ConfigurationClassEnhancer.java:358)
	at com.yisquare.ba.scanner.config.PointConfig$$EnhancerBySpringCGLIB$$c915987d.getPointsAndProperties(<generated>)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.springframework.beans.factory.support.SimpleInstantiationStrategy.instantiate(SimpleInstantiationStrategy.java:162)
	... 32 more

michelseiler, 2018-10-19:
You have to specify the string correctly.
AnalogInput is "analog-input".

`ObjectType forName = ObjectType.forName("analog-input");`

duyanhan1995, 2018-10-19:
@michelseiler   Ok, I'll try

duyanhan1995, 2018-10-20:
You successfully solved my problem, thank you @michelseiler 👍

closed 2018-10-20 by duyanhan1995