RadixIoT/BACnet4J #35: Maven issues with 5.0.0-SNAPSHOT and 4.1.7

gopalakrishnan-subramani, 2019-02-11:
Tried 5.0.0-SNAPSHOT as shown below, IntelliJ/Maven not able to find the version, similarly 

<dependency>
    <groupId>com.serotonin</groupId>
    <artifactId>bacnet4j</artifactId>
    <version>5.0.0-SNAPSHOT</version>
</dependency>

I have included repositories as shown in the document. 

<repositories>
    <repository>
..
..

4.1.7 also fails with maven, but 4.1.6 works. Any documentation or pointer to fix the issue really helpful.

terrypacker, 2019-02-11:
The available versions are:
4.1.6
4.1.7-SNAPSHOT
5.0.0-SNAPSHOT

Snapshots are not release ready so they should only be used for development.  The 5.0.0 release will probably be out in a month or so.

splatch, 2019-02-11:
> Snapshots are not release ready so they should only be used for development. The 5.0.0 release will probably be out in a month or so.

The [gradle script](https://github.com/infiniteautomation/BACnet4J/blob/master/build.gradle#L9) is already set to 5.0.0. I guess this might be reason for above misunderstanding. If current master is not production ready then it should remain with snapshot version per maven guidelines.

terrypacker, 2019-02-11:
Apologies for that, I use maven for the build and didn't notice the gradle script was wrong.  That is now fixed.

closed 2019-11-20 by terrypacker