RadixIoT/BACnet4J #39: Remove dependencies version range

Frozenlock, 2019-07-18:
Many of the dependencies have version range, such as `sero-scheduler`, `sero-warp`, etc.

This may cause repeatability problems as things can "shift under our feet" so to speak. It would be be preferable to avoid using ranges and declare a particular versions instead.

jazdw, 2019-07-18:
You can define them in your own pom.xml to fix them if you desire. But yes I tend to agree that we should change it.

mlohbihler, 2019-07-19:
Note that BACnet4J, sero-scheduler, and sero-warp were all written by the same developer, who i'm told isn't working on those libraries right now, so risk here is probably minimal. Also, sero-warp is only a test dependency, and i don't recall exactly but sero-scheduler may be too.

closed 2020-11-04 by jazdw