RadixIoT/BACnet4J #42: BACnet4j-5.0.0.jar with JAVA 8 is not supported in Android 7.0

SureshKumarSelvaraj, 2019-09-19:
I am Using BACnet4j-5.0.0.jar in my android application to create bacnet objects in my android app.

**To Reproduce**
Steps to reproduce the behavior:
1. Add BACnet4j-5.0.0.jar as library in android project
2. RUN in android 7.0 device
3. Getting Run time error 
"java.lang.NoClassDefFoundError: Failed resolution of: Ljava/time/Duration;"

Android 7.0 is not supporting all the features of JAVA 8.
Supports only the following
  Lambda expressions
  Method references
  Type annotations 
  Default and static interface methods 
  Repeating annotations 

Java.time is not supported by android 7.0. 

Suggest me some solution to make BACnet4j-5.0.0.jar compatible with android 7.0.

Thanks.

terrypacker, 2019-09-19:
This would not be easy.  

1. You could fork our repo and replace all references to `java.time` with whatever is supported on Android 7.0

2. Go back and see when in the history of BACnet4J was released for Java 7 or Java 6

closed 2019-09-19 by terrypacker

fozhao, 2021-06-18:
The release support Java 7 doesn't support protocol revision 14 or later. It's too old a revision to be honest. Latest BTL certification recommend at least revision 14. It makes this repo almost useless on old Android device.