RadixIoT/BACnet4J #43: The BTL compliance modifications shun non-conforming devices

Puckfist, 2019-10-03:
This thread brought it up, where they have a device that gives its protocol version as a string instead of an integer, which throws and exception in BACnet4J v5 and thus is not discovered: https://forum.infiniteautomation.com/topic/4380/es-version-bacnet-module-3-6-0-issue-compare-ver-3-5-2

Terry added this test to replicate specifically that occurrence: https://github.com/infiniteautomation/BACnet4J/commit/b4fedf61b462efa33a323ba96e121240ca0f8da8

This is the commit that added the exception in this specific case:  https://github.com/infiniteautomation/BACnet4J/commit/d8776ac276d201588630fc3bad37f3e6793e64c0#diff-2920871d456d6af7107fcca77b1fc622R188

I reached out to Michel Seiler several different ways but did hear back.

michelseiler, 2019-10-03:
I apologize for the circumstances.
The adjustment was due to the BTL test plan 15.1.final 4.6.1. This demands the ASHRAE standard test 9.22.2.3:

> Purpose: To verify that the IUT correctly responds to an attempt to write a property value that has an invalid datatype.
> Test Concept: The TD shall select an object in the IUT that contains a writable property designated P1. An attempt will be
> made to write to this property using a datatype that the IUT supports but which is invalid for the property. If no object
> supports writable properties, then this test shall be omitted.
> 
> Test Steps:
> 1. READ V = (Object1), P1
> 2. TRANSMIT WriteProperty-Request,
> 	'Object Identifier' = Object1,
> 	'Property Identifier' = P1,
> 	'Property Value' = (any value with an invalid datatype)
> 3. RECEIVE (BACnet-Error PDU,
> 	Error Class = PROPERTY,
> 	Error Code = INVALID_DATATYPE)
> 	| (BACnet-Reject-PDU
> 	Reject Reason = INVALID_PARAMETER_DATATYPE)
> 	| (BACnet-Reject-PDU
> 	Reject Reason = INVALID_TAG)
> 4. VERIFY (Object1), P1 = V

This test only describes the behavior when writing with an invalid datatype to our device,
not reading an invalid datatype on another device.
Maybe we can only check the datatype when writing to our device?

terrypacker, 2019-10-09:
Agreed, the check should only be made when writing to our device.  This puts the responsibility on the data consumer when reading.  I will make these changes and refer the commits to this issue.

michelseiler, 2019-10-11:
The same is true for the implementation of the test 9.22.2.4 (Writing with a Property Value that is Out of Range).
Again, the verification must be done only when writing to our device.
Its in the same commit: [d8776ac276d201588630fc3bad37f3e6793e64c0](https://github.com/infiniteautomation/BACnet4J/commit/d8776ac276d201588630fc3bad37f3e6793e64c0#diff-7d79ad6cf861cff4bf9533fec6d04b3fR51)

terrypacker, 2019-10-15:
@michelseiler I was hoping to get you to review this pull request or at least comment on it:

https://github.com/infiniteautomation/BACnet4J/pull/44

Thanks.

PR #44 (2019-10-16): Implement type validation for BTL certification at device write level
This is one possible solution that uses validation just prior to writing the value to the DeviceObject to fix #43
closed 2019-10-16 by terrypacker