RadixIoT/BACnet4J #45: Wrong object properties defined for weeklySchedule?

Frozenlock, 2019-11-08:
The `PropertyTypeDefinition` for [weeklySchedule](https://github.com/infiniteautomation/BACnet4J/blob/master/src/main/java/com/serotonin/bacnet4j/obj/ObjectProperties.java#L2282) appears to return inconsistent information (unless I misunderstood something):

`getArrayLength` -> 7

`isArray` -> false

Even the other sequence-like methods return false:
`isCollection` -> false
`isList` -> false

Shouldn't `isArray` return true whenever `getArrayLength` is larger than zero?

michelseiler, 2019-11-11:
According to BACnet Standard 135-2016, it should be a BACnetARRAY [7] of BACnetDailySchedule.

I get the correct output with my (not up to date) fork:
```
ObjectPropertyTypeDefinition objectPropertyTypeDefinition = ObjectProperties.getObjectPropertyTypeDefinition(ObjectType.schedule, PropertyIdentifier.weeklySchedule);
System.out.println("clazz: " + objectPropertyTypeDefinition.getPropertyTypeDefinition().getClazz());
System.out.println("isCollection: " + objectPropertyTypeDefinition.getPropertyTypeDefinition().isCollection());
System.out.println("isArray: " + objectPropertyTypeDefinition.getPropertyTypeDefinition().isArray());
System.out.println("isList: " + objectPropertyTypeDefinition.getPropertyTypeDefinition().isList());
System.out.println("arrayLength: " + objectPropertyTypeDefinition.getPropertyTypeDefinition().getArrayLength());

// Output:
// clazz: class com.serotonin.bacnet4j.type.constructed.DailySchedule
// isCollection: true
// isArray: true
// isList: false
// arrayLength: 7

```

michelseiler, 2019-11-13:
I have run the code with the master branch and get the same result.

Frozenlock, 2019-11-13:
Yes, you are correct.
I tried again, from scratch, and It was caused by a mistake on my part when retrieving the property.
My apologies.

closed 2019-11-13 by Frozenlock