RadixIoT/BACnet4J #46: Unable to get properties of remote device

lovefamilyC, 2019-12-02:
I obtained the remote VTS device through the code and it showed that the communication was successful, but when getting the device object property value, a timeout exception occurred. code show as below:
`public class TestMy {
    public static void main(String[] args) throws Exception {
        //创建一个虚拟设备 用于与设备交流  广播地址"255.255.255.255",端口47809
        IpNetwork network = new IpNetworkBuilder()
                .withLocalNetworkNumber(0)
                //第二从参数是子网掩码位数
                .withBroadcast("255.255.255.255",24)
                //.withBroadcast("192.168.1.255", 24)
                //.withSubnet("192.168.1.174",24)
                .withLocalBindAddress("192.168.1.174")
                //配置第一个网络，确保我们设置了重用地址
                .withPort(47808).withReuseAddress(true).build();

        Transport transport = new DefaultTransport(network);
        LocalDevice ld = new LocalDevice(1, transport);
        ld.initialize();
        //创建设备结束--------------
        ld.startRemoteDeviceDiscovery();

        //通过地址、设备号 获取远程设备属性
        RemoteDevice rd = ld.getRemoteDeviceBlocking(10);

        Map<PropertyIdentifier, Encodable> values = RequestUtils.getProperties(ld, rd, new ObjectIdentifier(ObjectType.device, 10), null,
                PropertyIdentifier.objectList);

        System.out.println(values);

        ld.terminate();

    }
}`

Error message：
`Exception in thread "main" com.serotonin.bacnet4j.exception.BACnetTimeoutException
	at com.serotonin.bacnet4j.transport.ServiceFutureImpl.result(ServiceFutureImpl.java:75)
	at com.serotonin.bacnet4j.transport.ServiceFutureImpl.get(ServiceFutureImpl.java:64)
	at com.serotonin.bacnet4j.util.RequestUtils.sendOneAtATime(RequestUtils.java:425)
	at com.serotonin.bacnet4j.util.RequestUtils.readProperties(RequestUtils.java:396)
	at com.serotonin.bacnet4j.util.RequestUtils.readProperties(RequestUtils.java:257)
	at com.serotonin.bacnet4j.util.RequestUtils.getProperties(RequestUtils.java:141)
	at com.serotonin.bacnet4j.util.RequestUtils.getProperties(RequestUtils.java:136)
	at com.serotonin.bacnet4j.TestMy.main(TestMy.java:46)`

yasar456, 2023-08-09:
Did you find the answer ?

terrypacker, 2023-08-09:
You need to bind to the wildcard address to receive UDP broadcast responses.  `0.0.0.0`

terrypacker, 2023-08-09:
Discovery does a WhoIs which is a broadcast request.  Its likely that you aren't seeing the responses.

closed 2025-12-03 by mlohbihler