RadixIoT/BACnet4J #48: Non-SNAPSHOT release needed for 5.0.2

WhittlesJr, 2019-12-20:
```
Caused by: org.eclipse.aether.resolution.ArtifactResolutionException: Could not transfer artifact com.serotonin:bacnet4j:pom:5.0.2 from/to ias-snapshots (https://maven.mangoautomation.net/repository/ias-snapshot/): Repository version policy: SNAPSHOT does not allow version: 5.0.2 (400)
```

terrypacker, 2019-12-20:
@WhittlesJr it looks like you are trying to use the ias-snapshots repo to get a release.  Use the ias-releases repo:

WhittlesJr, 2019-12-20:
I have both ias-releases and ias-snapshots configured. If I take away ias-snapshots, it fails to resolve 5.0.2:

`
Error building classpath. Could not find artifact com.serotonin:bacnet4j:jar:5.0.2 in central (https://repo1.maven.org/maven2/)
`

terrypacker, 2019-12-20:
Ah I see.  In 5.0 the maven coordinates changed:
```
	<groupId>com.infiniteautomation</groupId>
	<artifactId>bacnet4j</artifactId>
```

WhittlesJr, 2019-12-20:
Oh ok. Thank you!

closed 2019-12-20 by terrypacker