RadixIoT/BACnet4J #56: JSSC and Windows Crash

a-hansen, 2021-03-10:
One of our customers is trying to use MSTP on Windows and is experiencing the following:

[2021-02-02 11:19:53.928149] # A fatal error has been detected by the Java Runtime Environment:
[2021-02-02 11:19:53.928149] #
[2021-02-02 11:19:53.928149] #  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=1816, tid=0x000000000000035c
[2021-02-02 11:19:53.928149] #
[2021-02-02 11:19:53.928149] # JRE version: Java(TM) SE Runtime Environment (8.0_281-b09) (build 1.8.0_281-b09)
[2021-02-02 11:19:53.928149] # Java VM: Java HotSpot(TM) 64-Bit Server VM (25.281-b09 mixed mode windows-amd64 compressed oops)
[2021-02-02 11:19:53.928149] # Problematic frame:
[2021-02-02 11:19:53.943447] # C  [jSSC-2.8_x86_64.dll+0xb5db]
[2021-02-02 11:19:53.943447] #
[2021-02-02 11:19:53.943447] # Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
[2021-02-02 11:19:53.943447] #
[2021-02-02 11:19:53.943447] # An error report file with more information is saved as:
[2021-02-02 11:19:53.943447] # c:\dsa\dglux-server\dslinks\dslink-java-bacnet2\hs_err_pid1816.log
[2021-02-02 11:19:53.959071] #
[2021-02-02 11:19:53.959071] # If you would like to submit a bug report, please visit:
[2021-02-02 11:19:53.959071] #   http://bugreport.java.com/bugreport/crash.jsp
[2021-02-02 11:19:53.959071] # The crash happened outside the Java Virtual Machine in native code.
[2021-02-02 11:19:53.959071] # See problematic frame for where to report the bug.
[2021-02-02 11:19:53.959071] #
[2021-02-02 11:19:54.084071] == Exited with status 1 ==

It looks like JSSC is no longer maintained by the original developers.  However it has been forked here:

https://github.com/java-native/jssc

Using their latest rev fixed my customer's issue.  However, I don't really know if this version is safe to release for everyone.

You can find in the bug report about this issue where they (java-native maintainers) claim the original maintainers of JSSC helped them start the fork:

https://github.com/java-native/jssc/issues/65#issuecomment-662509700

Anyway, I'm reluctant to release this without bacnet4j adopting it as well.  Have you vetted the java-native fork, or are you willing to consider it?

Thanks,
Aaron

a-hansen, 2021-03-10:
I've done a little more digging.  There are many forks of JSSC but this looks one of the few actively maintained.  It's the top result on Google for "jssc github".  One of the [maintainers](https://github.com/bmarwell) is on the project management committee for Apache Shiro and is also a contributor to Maven.  It's not much but it makes me feel better.  I'm going to recommend we (DGLogik) release with this lib and be your guinea pig.  If I have any issues (and I remember) I'll update you.  Feel free to ping me as well.

splatch, 2021-03-10:
Serial libraries in java seem to be never ending drama. I know for sure that openHAB uses https://github.com/NeuronRobotics/nrjavaserial which is an active fork of ancient versions of RXTX. Seems there is a windows version. For mstp master all you need is serial input and output streams so you can try building mstp network from supplied streams:
```
    MasterNode node = new MasterNode(getSerialPortId(), port.getInputStream(), port.getOutputStream(), (byte)this.getStation(), 2);
    node.setMaxInfoFrames(5);
    node.setUsageTimeout(100);
    new MstpNetwork(node, 0);
```

I had this code in my bacnet binding but never actually tested it. Since I do not run windows on daily basis it is even harder for me to prove if it works. I will be testing this transport soon with linux. If you can, give a shot to nrjavaserial. Maybe it will solve your puzzle without major investments!