RadixIoT/BACnet4J #59: Cannot write null value to multistate object

envas, 2021-04-14:
**Describe the bug**
Commandable multistate objects do not accept NULL as a new property value. 

**To Reproduce**
1. Create a commanable multistate value object
2. Set some value for relinquish default
3. Write some valid value to level8 and override the default
4. Try to write NULL to level8 to turn back to default -> fires an error inconsistent configuration

**Expected behavior**
I can write NULL value to level8

IMO the problem is in the multistate mixin - line 55 - when validating the property. New value must be between 1 .. numStates but there is no test for NULL values:

https://github.com/infiniteautomation/BACnet4J/blob/6d34e6dd9e83a6fdea0708a72d0a46017386bbeb/src/main/java/com/serotonin/bacnet4j/obj/mixin/MultistateMixin.java#L52-L57

kishorevenki, 2021-04-15:
Writing to NULL is allowed if and only if the object supports Command Prioritization. In your application merely adding priority array would not make the object commandable. 
There should be line similar to  mv1.supportCommandable(new UnsignedInteger(1));
With the above line Commandablemixin  is activated and checking NULL @PRI is validated at CommandableMixin.java @line 250.  This is applicable to all commandable object types. Subsequently BACnet4J would function as intended.

Regards,
Kishore

envas, 2021-04-15:
I know about supportCommanable and command prioritization was supported. The problem was my application error, sorry, nothing to do with multistate or any other objects.

closed 2021-04-15 by envas