RadixIoT/BACnet4J #60: Error writing multistate property 'stateText'

envas, 2021-04-16:
**Describe the bug**
Cannot write multistate property `stateText`

**To Reproduce**
Steps to reproduce the behavior:
1. Create a MultistateValue object in a device
2. Connect a BACnet client and try to update stateText from the client
3. Error is generated in the device (server)

```
java.lang.ClassCastException: com.serotonin.bacnet4j.type.constructed.SequenceOf cannot be cast to com.serotonin.bacnet4j.type.constructed.BACnetArray
	at com.serotonin.bacnet4j.obj.mixin.MultistateMixin.validateProperty(MultistateMixin.java:63)
	at com.serotonin.bacnet4j.obj.BACnetObject.writeProperty(BACnetObject.java:481)
	at com.serotonin.bacnet4j.service.confirmed.WritePropertyRequest.handle(WritePropertyRequest.java:102)
	at com.serotonin.bacnet4j.transport.DefaultTransport.handleConfirmedRequest(DefaultTransport.java:870)
	at com.serotonin.bacnet4j.transport.DefaultTransport.incomingConfirmedRequest(DefaultTransport.java:827)
	at com.serotonin.bacnet4j.transport.DefaultTransport.receiveAPDU(DefaultTransport.java:640)
	at com.serotonin.bacnet4j.transport.DefaultTransport.receiveImpl(DefaultTransport.java:578)
	at com.serotonin.bacnet4j.transport.DefaultTransport.run(DefaultTransport.java:498)
	at java.lang.Thread.run(Thread.java:748)
```
**Expected behavior**
I can update stateText from a BACnet client

I logged the request in 'DeviceEventAdapter.allowPropertyWrite()' listener callback and the Property Value is really of the type SequenceOf. The log from my device (server):

```
2021-04-16 15:09:03,140 - BACnet write property write request received from=192.168.0.130:0xBAC0
2021-04-16 15:09:03,140 -          object-name  : 71L2Az00_plant_mod
2021-04-16 15:09:03,141 -          object-type  : multi-state-value
2021-04-16 15:09:03,141 -          property     : state-text
2021-04-16 15:09:03,141 -          value        : [Anlage Aus, ON-Optimum Start, Ein-Stützbetrieb, Ein-Nachtkühlung, Ein-Zeitprogramm, Ein-Übersteuerung, Ein-Nachlauf]
2021-04-16 15:09:03,141 -          value-type   : SequenceOf
```

mlohbihler, 2025-12-03:
According to the spec the "State_Text" property of a multistate object is a `BACnetArray`, not a `SequenceOf`. What is the client you are using? Perhaps it is sending the incorrect object type in the request?

mlohbihler, 2025-12-04:
> I have written a simple BACnet Client using BACnet4J (and JavaFX as GUI builder). Now both client and server are using BACnet4J. When my client reads Multi-State Value Object, the type of the property State-Text is SequenceOf. I am pretty sure it must be an issue in BACnet4J.

I agree. I have recreated the issue myself, and i think i have a fix. I just need to write unit tests for it, and handle some internal issue tracking paperwork. But hopefully this can be included in the next release.