RadixIoT/BACnet4J #62: How to use this with routed connectivity with bacnet?

danishmhmd, 2021-08-04:
Please help me using this library with a routed network to bacnet device, currently my device is not connected to bacnet directly on any interface but rather it is routed via a managed switch in which bacnet and my device is on a completely different network, the packets are routed via network routing done in switch. If i send UDP packets directly it works but i am unable to see any devices by sending a who is request. Below is the sample command i sent and response i received from bacnet 

Sending Command to Bacnet : 81 a 0 15 1 4 2 75 7c 5 9 1 1c 4 c0 0 1 29 0 39 78 
Response from Bacnet :  81 a 0 9 1 0 60 7c 9

My interface ip:        inet 192.168.1.200  netmask 255.255.255.0  broadcast 192.168.1.255
Bacnet ip : 192.168.2.200

logs which i see after sending who is:-
[main] DEBUG com.serotonin.bacnet4j.transport.DefaultTransport - Broadcasting WhoIsRouter to local network
[main] DEBUG com.serotonin.bacnet4j.util.RemoteDeviceFinder - Waiting 10000 ms for something to happen
[main] DEBUG com.serotonin.bacnet4j.util.RemoteDeviceFinder - Done waiting
[main] DEBUG com.serotonin.bacnet4j.util.RemoteDeviceFinder - Throwing timeout

kishorevenki, 2021-08-05:
Any router or Layer 3 Managed switch would never allow any broadcast messages from one subnet to another. They will  allow only Unicast messages. Because of the same Who-Is from 192.168.1.200 would never reach to 192.168.2.200. 

To accomplish this you have to configure both the devices as BBMD device. Once BBMD are configured you need to add the IP address of other BBMD devices and self in the BDT table of each device. Once they are configured, communuication would happen.

danishmhmd, 2021-08-05:
Thanks this would help

closed 2021-08-05 by danishmhmd