RadixIoT/BACnet4J #63: COV not working with multiStateValue,multiStateOutput,multiStateInput

danishmhmd, 2021-08-05:
Hi I was exploring the library for all different COV type but i noticed i am unable to get the COV packets on my listener, however i checked wireshark there seems to be some error of unknown service while subscribing even after that device do sends the notification of which library reply with unknown service. However same is working with a third party online tool i checked called Yabe. PFA screenshot of wire shark with bacnet4j and yabe. Below are the logs which i see when a notification is sent from device:
[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from local network. From=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], local=0
[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from [a,0,0,66,ba,c0]: NPDU [from=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], linkService=null, queue=[0,4,f6,1,9,4,1c,2,0,1e,2c,2c,4,c0,0,5,39,0,4e,9,55,2e,91,3,2f,9,6f,2e,82,4,0,2f,4f]]
[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from local network. From=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], local=0
[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from [a,0,0,66,ba,c0]: NPDU [from=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], linkService=null, queue=[0,4,f7,1,9,1,1c,2,0,1e,2c,2c,4,c0,0,5,39,0,4e,9,55,2e,91,3,2f,9,6f,2e,82,4,0,2f,4f]]
[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from local network. From=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], local=0
[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from [a,0,0,66,ba,c0]: NPDU [from=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], linkService=null, queue=[0,4,f8,1,9,4,1c,2,0,1e,2c,2c,4,c0,0,5,39,0,4e,9,55,2e,91,0,2f,9,6f,2e,82,4,0,2f,4f]]
[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from local network. From=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], local=0
[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from [a,0,0,66,ba,c0]: NPDU [from=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], linkService=null, queue=[0,4,f9,1,9,1,1c,2,0,1e,2c,2c,4,c0,0,5,39,0,4e,9,55,2e,91,0,2f,9,6f,2e,82,4,0,2f,4f]]
[image]
[image]

kishorevenki, 2021-08-05:
Have you set COVNotification to TRUE like 
servicesSupported.setConfirmedCovNotification(true);

danishmhmd, 2021-08-06:
Yes as below:
remoteDevice.getServicesSupported().setConfirmedCovNotification(true);
Here are the console logs of library when i device sends the COV and library reply it with service not supported:-

[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from local network. From=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], local=0
[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from [a,0,0,66,ba,c0]: NPDU [from=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], linkService=null, queue=[0,4,a8,1,9,1,1c,2,0,1e,2c,2c,4,c0,0,5,39,0,4e,9,55,2e,91,0,2f,9,6f,2e,82,4,0,2f,4f]]

kishorevenki, 2021-08-07:
After carefully analyzing the wireshark logs sent earlier I noticed the following:
1. 10.0.0.100 - BACnet4J as workstation, 10.0.0.102 - Controller having AI,0 and MSV,5.
2. 10.0.0.100 subscribes COV for both AI,0 and MSV,5 which has been accepted by the controller as well.
3. Subsequently AI,0 sent COV notification which was accepted by BAcnet4J. 
4. But when the COV is received from MSV,5 it throws error. 
5. So the issue is related to MSV and not with COV. 

So please look at the code around MSV and if possible share the code so as to reproduce from my side to see how can I help you.

danishmhmd, 2021-08-07:
Sure, Please find the sample code snippet below for communication class and listener :-

=================================================================================================

			IpNetworkBuilder builder = new IpNetworkBuilder();
			builder.withPort(Variables.port);
			builder.withSubnet(Variables.subnetAddress, Variables.networkPrefixLength);
			builder.withReuseAddress(true);
			IpNetwork network = builder.build();
			Transport transport = new DefaultTransport(network);
                        LocalDevice localDevice;
			localDevice = new LocalDevice(Variables.device_no, transport);
			localDevice.initialize();
                        localDevice.getEventHandler().addListener(new CustomListener(localDevice));

 WhoISRequestCustom whois = new WhoISRequestCustom(localDevice);
                whois.whois();

                       SubscribeCOVRequest subscribeCOVRequest = new SubscribeCOVRequest(new UnsignedInteger(1),                       new ObjectIdentifier(ObjectType.multiStateValue, 5),
                     Boolean.TRUE, new UnsignedInteger(0));

                     SubscribeCOVRequest subscribeCOVRequest2 = new SubscribeCOVRequest(new UnsignedInteger(4),
                     new ObjectIdentifier(ObjectType.analogInput, 0),
                     Boolean.TRUE, new UnsignedInteger(0));

                     localDevice.send(remoteDevice1,
                     subscribeCOVRequest).get();
                     localDevice.send(remoteDevice1,
                     subscribeCOVRequest2).get();

======================================================================

public class CustomListener extends DeviceEventAdapter
{
   LocalDevice localDevice;

   public CustomListener(LocalDevice localDevice)
   {
      this.localDevice = localDevice;
   }

   @Override
   public void iAmReceived(final RemoteDevice d)
   {
      System.out.println("I AM FOR Devcice " + d.toString());
      new Thread(new Runnable() {
         public void run()
         {
            try
            {
               DiscoveryUtils.getExtendedDeviceInformation(localDevice, d);
               SequenceOf<ObjectIdentifier> oids =
                        RequestUtils.getObjectList(localDevice, d);

               System.out.println("I AM " + oids);

            }
            catch (BACnetException e)
            {
               e.printStackTrace();
            }
         }
      }).start();
   }

   public void covNotificationReceived(
            final UnsignedInteger subscriberProcessIdentifier,
            final ObjectIdentifier initiatingDeviceIdentifier,
            final ObjectIdentifier monitoredObjectIdentifier,
            final UnsignedInteger timeRemaining,
[... 93 lines cut ...]
                  + service.toString());
      }
      catch (Exception e)
      {
         // TODO: handle exception
      }
   }

   @Override
   public void textMessageReceived(
            final ObjectIdentifier textMessageSourceDevice,
            final Choice messageClass, final MessagePriority messagePriority,
            final CharacterString message)
   {
      System.out.println("Test message ");
   }

   @Override
   public void synchronizeTime(final Address from, final DateTime dateTime,
            final boolean utc)
   {
      System.out.println("synchronizeTime ");
   }

}

danishmhmd, 2021-08-07:
WireShark Capture attached as well
[Capure.pcapng.txt](https://github.com/infiniteautomation/BACnet4J/files/6948765/Capure.pcapng.txt)

kishorevenki, 2021-08-09:
[image]
I tried the code, but I receicved Simple Ack for MSV,1 and AI,1. Screenshot of the wireshark is enclosed.  

192.168.1.143  - BACnet4J

danishmhmd, 2021-08-11:
Sure, but are you getting these packets in the listener you are using on this remote object in the method covNotificationReceived()?
The problem is in some parsing of the COV notification, This is the error which i see on console upon setting issueConfirmedNotifications to false by below, in this case there is no error code sent to bacnet just the parsing error in packet class:

SubscribeCOVRequest subscribeCOVRequest = new SubscribeCOVRequest(new UnsignedInteger(1),
                     new ObjectIdentifier(ObjectType.multiStateValue, 5),
                     Boolean.TRUE, new UnsignedInteger(0));

[BACnet4J IP socket listener for device 0] DEBUG com.serotonin.bacnet4j.npdu.Network - Received NPDU from [a,0,0,66,ba,c0]: NPDU [from=Address [networkNumber=0, macAddress=[a,0,0,66,ba,c0]], linkService=null, queue=[10,2,9,1,1c,2,0,1e,2c,2c,4,c0,0,5,39,0,4e,9,55,2e,91,0,2f,9,6f,2e,82,4,0,2f,4f]]
[BACnet4J transport for device 0] ERROR com.serotonin.bacnet4j.event.DefaultExceptionListener - 
com.serotonin.bacnet4j.exception.BACnetErrorException: property: invalid-data-type
	at com.serotonin.bacnet4j.type.primitive.Primitive.readTag(Primitive.java:188)
	at com.serotonin.bacnet4j.type.primitive.UnsignedInteger.<init>(UnsignedInteger.java:107)
	at sun.reflect.GeneratedConstructorAccessor1.newInstance(Unknown Source)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at com.serotonin.bacnet4j.type.Encodable.read(Encodable.java:230)
	at com.serotonin.bacnet4j.type.AmbiguousValue.convertTo(AmbiguousValue.java:180)
	at com.serotonin.bacnet4j.type.Encodable.readANY(Encodable.java:440)
	at com.serotonin.bacnet4j.type.constructed.PropertyValue.<init>(PropertyValue.java:84)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at com.serotonin.bacnet4j.type.Encodable.read(Encodable.java:230)
	at com.serotonin.bacnet4j.type.constructed.SequenceOf.<init>(SequenceOf.java:87)
	at com.serotonin.bacnet4j.type.Encodable.readSequenceOf(Encodable.java:309)
	at com.serotonin.bacnet4j.service.unconfirmed.UnconfirmedCovNotificationRequest.<init>(UnconfirmedCovNotificationRequest.java:88)
	at com.serotonin.bacnet4j.service.unconfirmed.UnconfirmedRequestService.createUnconfirmedRequestService(UnconfirmedRequestService.java:80)
	at com.serotonin.bacnet4j.apdu.UnconfirmedRequest.parseServiceData(UnconfirmedRequest.java:89)
	at com.serotonin.bacnet4j.transport.DefaultTransport.receiveAPDU(DefaultTransport.java:646)
	at com.serotonin.bacnet4j.transport.DefaultTransport.receiveImpl(DefaultTransport.java:578)
	at com.serotonin.bacnet4j.transport.DefaultTransport.run(DefaultTransport.java:498)
	at java.lang.Thread.run(Thread.java:748)

danishmhmd, 2021-08-11:
I guess i found the issue i was using version 5.0.0 and you are testing on 6.0.0-SNAPSHOT. Let me once check this with version and update back