RadixIoT/BACnet4J #65: maven build is failing

coolsan24, 2021-12-02:
Failed to read artifact descriptor for com.serotonin:bacnet4j:jar:4.1.6: Could not transfer artifact com.serotonin:bacnet4j:pom:4.1.6 from/to ias-releases (https://maven.mangoautomation.net/repository/ias-release/): PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed: NotAfter: Wed Dec 01 23:59:59 UTC 2021 -> [Help 1]

jinuxx, 2021-12-03:
same here, seems that the maven repo is broken.

coolsan24, 2021-12-03:
yeah , I think they need to fix it soon ,... anybody can you check this please

vmustafayev4Eqnx, 2021-12-03:
Hi, seems like the certificate expired. Could anyone fix it ASAP?
Thanks in advance

mlohbihler, 2025-12-03:
Closing due to lack of activity