RadixIoT/BACnet4J #66: BACnet message is processing infinitely

lubomir-mitev, 2021-12-13:
BACnet version - 6.0.0. 

We found that a message received from the bacnet4j leads to it's infinitely processing. This mean high CPU and no more communication with the devices. This is the log information:
Received NPDU from [a,23,d4,e6,ba,c2]: NPDU [from=Address [networkNumber=0, macAddress=[a,23,d4,f,ba,c2]], linkService=null, queue=[10,4,9,7,1a,2,2,2e,0,0,0,5,0,0,0,20,1,b0,c8,6,0,0,0,2,0,3,a9,84,0,0,0,b,5e,d,9f,83,0,54,43,ae,0,0,0,7,f,ae,1,b0,38,3,10,0,0,0,40,45,ff,f0,0,f,b7,1,b0,50,2,10,0,0,0,20,0,0,0,1,f,fb,1,b0,50,3,10,0,0,0,40,42,87,0,0,f,fc,1,b0,50,4,10,0,0,0,20,0,0,0,3,f,fd,1,b0,50,5,10,0,0,0,40,42,c8,0,0,f,fe,1,b0,50,6,10,0,0,0,20,0,0,0,7,f,a0,1,b0,e8,d,10,0,0,0,20,0,0,0,3,2f]]

We created simple test - APDU with this queue. It's UnconfirmedRequest and when DefaultTransport invokes the method parseServiceData(), it never finish processing the data.

I don't know the BACnet device and also I don't know if the payload is correct or not. But the library must not process a message infinitely and must not stop the communications with the devices.

terrypacker, 2021-12-13:
@lubomir-mitev can you provide your test code here?  That will save time and I can review the problem and provide a fix.

Thanks for letting us know about this.

lubomir-mitev, 2021-12-14:
[InfinitelyTest.txt](https://github.com/MangoAutomation/BACnet4J/files/7709342/InfinitelyTest.txt)
File is attached. Rename the file extension to java, because It's not possible to attach java file :( It's a unit test.

Thisora, 2024-12-05:
I see that you set only setUnconfirmedPrivateTransfer to true.
I suppose that you are testing this kind of message.

I have an ArrayIndexOutOfBoundsException that is throw by ByteQueue.