RadixIoT/BACnet4J #68: Impossible to configure BBMD?

Frozenlock, 2022-04-06:
As far as I can tell, it's impossible to configure the BBMD table of a local device with methods/functions.

Looking [at the tests](https://github.com/MangoAutomation/BACnet4J/blob/141fa420aa6bc1897dbc94c2c504558d98bdc6a9/src/test/java/com/serotonin/bacnet4j/npdu/ip/BBMDTest.java#L219), it looks like a remote device is simulated in order to to configure each BBMD.

Is my understanding correct? 
Which approach should I take if I want to create a BBMD table by providing multiple hosts/ports?

envas, 2023-03-15:
Yes, you understand well. BBMD is completely unusable with BACnet4J. If one wants to program a BBMD router, one must be able to populate the BDT table with data. This is typically done when restarting a BBMD router. How the BDT content is distributed across all BBMD routers is not specified in the BACnet specification. The BACnet best practice only says that the BDT content should be the same on all routers. Companies implement this in different ways - a database table, an INI file, a central server from which the router downloads data, there are various options. But they all have one thing in common - the BBMD router must be able to populate the contents of the BDT table after a reboot. This is not possible with BACnet4J, because all API functions for content manipulation (IpNetwork.readBDT and IpNetwork.writeBDT) are private and can only be called via BACnet services.

Frozenlock, 2023-03-15:
Thank you for confirming my suspicions and providing a detailed answer!

terrypacker, 2023-03-15:
@Frozenlock if you have any suggestions for us to expose the methods you require let us know.  A simple PR would be helpful.

envas, 2023-03-16:
@kishorevenki thank you for the detailed explanation. I am developer, not engineer, and I'm not talking about commissioning. I'm looking for a way how to populate the BDT on **configured** BBMD devices after a reboot, when the BDT content was cleared. BACnet4J doesn't support NetworkPort object, doesn't have local API to write to BDT, and doesn't even insert its own IP address into BDT after initialisation. So after a reboot/power failure I get an empty BDT. And there is no method to reconstruct the content of the BDT locally. The only way is to use remote device and send write-broadcast-distribution-table message from outside. Which is practically nonsense, because I would have to have some kind of watchdog that would permanently scan the entire distributed network and immediately remotely reloads the BBMD router with configuration data when it restarts.  

I see absolutely no reason why it would be dangerous to change the modifier of the Java methods `readBDT` and `writeBDT` from private to public. These methods are intended for programmers and cannot be used other than in an application.

envas, 2023-03-16:
@terrypacker Yes, I try to develop a BBMD device. This device has a web user interface, where I perform the commissioning. The commissioning form has a `Save` button, that persists the BDT content into an internal database. After reboot, I need to populate the local BDT content from the database to restore the functionality. And I do not see any API method how to write the BDT content. The BBMD test unit uses a remote device simulation to write into the BDT. I find it impractical to create a new virtual device just for this function.

I see that the BBMD basically works, I can register my device as a foreign device with a BBMD (what you probably means that Mango uses). Other direction works as well - I can activate BBMD after calling `network.enableBBMD()` and other devices can register on my device as foreign devices.  

My question is how to reload the BDT content after device reboot, like this (symbolic code)

```Java
IpNetwork network = new IpNetworkBuilder().withXXXX.withYYYY.build();
DefaultTransport transport = new DefaultTransport(network);
LocalDevice device = new LocalDevice(device_id, transport);
// enable BBMD 
network.enableBBMD();

// once enabled, others can write BDT (tested from Yabe and BacEye), however, assuming
// the BDT content was already configured, I need to reload it from the database
readDatabaseAndWriteBDT();    // <- that's what I am looking for

// configure other device properties and initialize device
device.initialize();
```

terrypacker, 2023-03-16:
Ok, yes that makes sense.  Are you comfortable forking this repo and making the changes that you require?  If so you could then make a Pull Request and we could verify and merge your changes back in.  At the very least it would be a starting point for us to add this feature.

terrypacker, 2023-03-16:
Ideally if you make the changes can you provide a test showing us how you would use this feature, that will give us a better understanding of why you want it and if there are other changes that would help you achieve this.

envas, 2023-03-16:
OK, give me a bit time, I have another Pull Requests in the queue. I will finish my project and then make a summary.

kishorevenki, 2023-03-17:
@envas - Now I understand your concerned which is valid.

devaskim, 2023-08-08:
Is there any plans to let configuring BDT with public library's API ?

mlohbihler, 2025-12-03:
Note that the issue of receiving broadcasts on non-wildcard addresses has been addressed: https://github.com/RadixIoT/BACnet4J/pull/134

mlohbihler, 2026-01-08:
> Note you should add local bind address to the BDT as first entry to enable passing forwarded messages to the local device because the local device can act as BBMD router and a controller simultaneously. This was not considered at all in the current implementation of BACnet4J BBMD.

The problem with such a default is that the distribution mask may not be correct. I think in any case the entire BDT should be rewritten by some configuration agent. Regardless, i have included this in my recent changes. I think this argely covers what has been discussed, and also addresses some issues with wildcard bind addresses.

https://github.com/RadixIoT/BACnet4J/pull/150

closed 2026-01-08 by mlohbihler