RadixIoT/BACnet4J #79: I wrote a program with bacnet4j. The jar package of the window platform can be scanned to the device when running. The device cannot be scanned on Linux Centos. They are all in the same network segment

youngKizZz, 2023-02-10:


terrypacker, 2023-02-10:
It would be helpful if you provide more specific information.  However I would guess that the "scan" you are talking about is the Discovery process.  This uses UDP broadcast packets, Windows and Linux handle these differently.  In linux if you don't bind to 0.0.0.0 you won't receive these messages.

abbottyf, 2023-04-25:
this is important  In linux if you don't bind to 0.0.0.0 you won't receive these messages.    First of all, you judge whether you are in the same network segment. You can only scan the device in the same network segment. If you are not in the same network segment, you can only cross network segments. Specify object-type and ins to obtain data.  Hope it helps you

terrypacker, 2023-04-25:
@wangdong-ivymobile, @abbottyf if correct.  Here is an old issue for the same problem:

https://github.com/MangoAutomation/BACnet4J/issues/53

wangdong-ivymobile, 2023-04-28:
@terrypacker @abbottyf thanks for the explanation.
I was really surprised by the behavior. Can we detect this problem and handle it in the library? Is there a better way other than binding to 0.0.0.0.
For example, I found some bacnet device broadcast to 255.255.255.255, some to 192.168.2.255(assuming 192.168.0.1/24), if we explicitly listen to broadcast channels, should we listen to both?

terrypacker, 2023-04-28:
I had played around with this a little some time ago and wrote this test:

https://github.com/MangoAutomation/BACnet4J/blob/master/src/test/java/com/serotonin/bacnet4j/adhoc/MultipleLocalDevices.java

qinyi8888, 2024-04-12:
1. Bind the ip address to: "0.0.0.0"
2. check the network settings, see if the network PREFIX, NETMASK of linux system and Windows computer are the same.
3. check if linux system firewall setting is normal, open udp port. You can try to close the firewall to try.
Hope it can help you.
[image]