RadixIoT/BACnet4J #81: BACnet error:invalid data-type

Pandacli, 2023-04-10:
Hi，all . I use  writePropertyRequest to write a new presentValue  , but unfortunately it response a wrong code 

invalid data-type. could you give us any advise?  thanks 

there is my code 
` WritePropertyRequest writePropertyRequest = new WritePropertyRequest(
                        objectList.get(i),
                        PropertyIdentifier.presentValue,
                        null, new Real(setVal), new UnsignedInteger(BaConfig.priority));
                ObjectType objectType = writePropertyRequest.getObjectIdentifier().getObjectType();
                ServiceFuture send = localDevice.send(remoteDevice, writePropertyRequest);

                ReadPropertyAck ack = localDevice.send(remoteDevice, writePropertyRequest)
                        .get();`
【project details 】
setValue = 1;
priority is 9 ; 

Device detail like this 
[image]

kishorevenki, 2023-04-11:
HI,
Based on your above information, I noticed couple of points:

1. Since the Present_Value is not of array datatype, this value can be omitted. Mentioning Null is not correct. 
2. Since the Object Type is Binary Value, the datatype of PV is not Real, but BinaryPV. Hence the value can be new BinaryPV(1). 

Regards,
Kishore

Pandacli, 2023-04-11:
Hi Kishore ,thanks for your advise 
and we fix it to change  the BinaryPV ,not a Real.

> HI, Based on your above information, I noticed couple of points:
>
> 1. Since the Present_Value is not of array datatype, this value can be omitted. Mentioning Null is not correct.
> 2. Since the Object [...]

kishorevenki, 2023-04-11:
Hope you will remove Null field as well. 

WritePropertyRequest writePropertyRequest = new WritePropertyRequest(
objectList.get(i),
PropertyIdentifier.presentValue,
new BinaryPV(setVal), new UnsignedInteger(BaConfig.priority));

Pandacli, 2023-04-11:
> Hope you will remove Null field as well.
> 
> WritePropertyRequest writePropertyRequest = new WritePropertyRequest( objectList.get(i), PropertyIdentifier.presentValue, new BinaryPV(setVal), new UnsignedInteger(BaConfig.priority));

Yes,it work . I should use BinaryPV.active to control  Bacnet device . thank you so much