RadixIoT/BACnet4J #84: BACnet Schedule

yasodakrishnav, 2023-07-20:
Hey Team,

I have created a schedule object using bacnet4j library in java. Like below

```
final AnalogValueObject av0 = new AnalogValueObject(d1, 100, "av0", 98, EngineeringUnits.amperes, false)
                .supportCommandable(-2);
final SequenceOf<CalendarEntry> dateList = new SequenceOf<>( 
                new CalendarEntry(new Date(-1, null, -1, DayOfWeek.FRIDAY)), // Every Friday.
                new CalendarEntry(new DateRange(new Date(-1, Month.NOVEMBER, -1, null), new Date(-1, Month.FEBRUARY, -1, null))), 
                new CalendarEntry(new WeekNDay(Month.FEBRUARY, WeekNDay.WeekOfMonth.days22to28, DayOfWeek.WEDNESDAY)));
final CalendarObject co = new CalendarObject(d1, 9999, "cal0", dateList);
final DateRange effectivePeriod = new DateRange(Date.UNSPECIFIED, Date.UNSPECIFIED);

        final BACnetArray<DailySchedule> weeklySchedule = new BACnetArray<>( //
                new DailySchedule(new SequenceOf<>(new TimeValue(new Time(8, 0, 0, 0), new Real(10)),
                        new TimeValue(new Time(17, 0, 0, 0), new Real(11)),
                        new TimeValue(new Time(19, 0, 0, 0), new Real(1112)))), //
                new DailySchedule(new SequenceOf<TimeValue>()), //
                new DailySchedule(new SequenceOf<TimeValue>()), //
                new DailySchedule(new SequenceOf<TimeValue>()), //
                new DailySchedule(new SequenceOf<TimeValue>()), //
                new DailySchedule(new SequenceOf<TimeValue>()), //
                new DailySchedule(new SequenceOf<TimeValue>()));

final SequenceOf<DeviceObjectPropertyReference> listOfObjectPropertyReferences = new SequenceOf<>(
                new DeviceObjectPropertyReference(av0.getId(), PropertyIdentifier.presentValue, null, null));

final ScheduleObject so = new ScheduleObject(d1, 10, "sch0", effectivePeriod, weeklySchedule, exceptionSchedule,
                new Real(21), listOfObjectPropertyReferences, 12, false);

```

Now the question is, I want set the null after Time(19, 0, 0, 0) on monday like below, but i couldn't able to do that, because  i'm getting the error  saying  **class=property, code=invalid-data-type
	at com.serotonin.bacnet4j.obj.ScheduleObject.validateProperty(ScheduleObject.java:191)**
```
new DailySchedule(new SequenceOf<>(new TimeValue(new Time(8, 0, 0, 0), new Real(10)),
                        new TimeValue(new Time(17, 0, 0, 0), new Real(11)),
                        new TimeValue(new Time(19, 0, 0, 0), new Real(1112))))
                        new TimeValue(new Time(20, 0, 0, 0), new Null())))
```

I want to set null value on Time(20, 0, 0, 0), because i want the default value to set it as present value.

Thanks

splatch, 2023-07-20:
I have not read spec since a while, hence I just guess that default value might be something which apply to situations when you miss a schedule entry for a given day (lets say Sunday). If you have a schedule for a given day the last time value defines what "rest of the day" might be. Was you able to read null value for time value in daily schedule?

kishorevenki, 2023-07-21:
NULL shall be allowed to configure in the value (time-value) of Daily Schedule, Exception Schedule and Schedule Default irrespective of whether the Schedule is of BinaryPV, or Real or Unsigned etc type.

yasodakrishnav, 2023-07-21:
[image]

Yeah they should allow null value, However, in the ScheduleObject.java class, specifically at line number 190, the time-value class should align with the default value class.

yasodakrishnav, 2023-07-21:
> I have not read spec since a while, hence I just guess that default value might be something which apply to situations when you miss a schedule entry for a given day (lets say Sunday). If you have a [...]

he purpose of allowing null values in the daily schedule is to ensure that if a null value is set in the weekly schedule with a certain priority, the present value will revert to its default value.

closed 2024-06-27 by MertCingoz