RadixIoT/BACnet4J #95: Bacnet4J cannot write BinaryPV properly

yangshuyi, 2024-06-13:
I'm now using Bacnet4j version 6.0.0.
I read the previous value in DDC, and try to write the new value, then I read from bacnet in a loop.

Below code can be executed without any error/exception, but the value still shows to be the previous one. Can anyone help me to how to deal with that?

Encodable writeValue = BinaryPV.active;

        RequestUtils.writeProperty(
                localDevice,
                remoteDevice,
                new ObjectIdentifier(ObjectType.binaryValue, objectInstance),
                PropertyIdentifier.presentValue,
                writeValue);

the output like:

previous value inactive
write: active
value:0
value:0
value:0
value:0
value:0
value:0
value:0
value:0
value:0

kishorevenki, 2024-06-14:
This behavior could be due to inactive value is set at some higher priority of BV object. Please read the priority-array property of BV that you are writing. If there is any value at higher priority other than NULL, then relinquish them all one-by-one by wiriting Present-Value to NULL in the appropriate priority.

yangshuyi, 2024-06-14:
you are correct. By setting all priority value to be null, then set my new value can take effect. Thank you.

By the way, do you know how to read out all priority-array values? 
There's only one API can read the value, but the last one does not mean priority.
```
public static Encodable readProperty(final LocalDevice localDevice, final RemoteDevice d,
            final ObjectIdentifier oid, final PropertyIdentifier pid, final UnsignedInteger propertyArrayIndex)
```

kishorevenki, 2024-06-14:
Good to know that your problem is resolved. In this API, PropertyArrayIndex is optional. So if you send RP request with oid = BV,0; pid = priority-array, the result will be array of 16 . 

Regards,
Kishore

yangshuyi, 2024-06-19:
I appreciate your kindly help. Thank you so much.

closed 2024-06-19 by yangshuyi