RadixIoT/BACnet4J #99: No response when execute RequestUtils.readProperties

yangshuyi, 2024-07-05:
I'm now using Polling way to fetch the value from device. Sometime when executing RequestUtils.readProperties, it will not come back any response (more than 30 seconds no response).  Normally, it will give back the value within 1second.

I have to wrap the invocation with a Future and set a timeout exception logic. But actually, I don't want to get any failure. 

Can someone kindly show me what's the reason.

        <dependency>
            <groupId>com.infiniteautomation</groupId>
            <artifactId>bacnet4j</artifactId>
            <version>6.0.0</version>
        </dependency>

kishorevenki, 2024-07-11:
Which is the device having address 192.168.30.255? This address looks like broadcast address. So any device that receives with the broadcast address would discard the packets, if the those mesages are not a valid broadcast message.  RP, WP, requests are unicast message and Who-Is, I-Am, Who-Has, I-have etc. are broadcast messages.

yangshuyi, 2024-07-11:
> Which is the device having address 192.168.30.255? This address looks like broadcast address. So any device that receives with the broadcast address would discard the packets, if the those mesages [...]

Thank you for answering the question. But why sometimes can get the packets, sometimes cannot?

kishorevenki, 2024-07-11:
Can you please share the wireshark logs?