RadixIoT/BACnet4J #103: How to get source code related to each release

fred-hartman, 2024-10-05:
New releases appear on MVN Repository only as a jar without the corresponding source code. Since 4.x releases (that contained the pair of binary+source), releases no longer appear in the releases list here in github.  We need to be able to periodically review the source and make sure it matches the binary of anything that touches production. And I assume as fear of supply chain attacks grow, this will become more widespread.  Also, we recently upgraded(5.0.2 to 6.0.0) and a production server is hanging multiple times a day.

Is it possible to provide the exact source snapshot that corresponds to each release as 4.x releases did? Even better is a branch or tag named for the release name.

This provides a way to quickly understand what has changed from one release to another for troubleshooting and allows us (or a trusted 3rd party) to build from that source snapshot to make sure the source creates the release binaries and scan that source.

I appreciate your ongoing work maintaining and enhancing BACnet4j.

terrypacker, 2024-10-07:
@fred-hartman the source is attached to the binary artifact in our sonatype repository here:

https://maven.mangoautomation.net/#browse/search=keyword%3DBACnet4j:NX.coreui.model.Component-10

I checked and the source seems to be attached to each release there.

fred-hartman, 2024-10-07:
Thank You!

These are the paths I tried:
1. MVN Repository has https://maven.mangoautomation.net/repository/ias-release/com/infiniteautomation/bacnet4j/6.0.0/, which is broken and following the Browse link to https://maven.mangoautomation.net/#browse/browse:ias-release is a tree that does not contain the recent bacnet releases.
2. The https://github.com/MangoAutomation/BACnet4J/releases link.

Is there a way to get to the 6.0.1 release?

terrypacker, 2024-10-07:
Interesting, yes you are correct I can't find it by browsing.  I was able to find it by searching for "BACnet4J"

[image]

There is no release for 6.0.1 as it is still in a `SNAPSHOT` development phase.

Can you find what you need by searching the Maven repository?

fred-hartman, 2024-10-07:
Yes. I found by searching . I didn't know 6.0.1 was not released yet. I'll wait. Thank you.

terrypacker, 2024-10-07:
No worries.  I'll make some notes on this to improve the way we tag releases and distribute them.

closed 2024-10-07 by terrypacker