RadixIoT/BACnet4J #105: Enhancement Request for Foreign Device Registration in `IpNetwork`

Thisora, 2024-11-27:
Hello,

First of all, thank you for your work on this library—it's been a great help!

I'm currently implementing a LocalDevice that needs to register itself as a foreign device on a remote BBMD (via FDT). Here’s a snippet of my code to provide context:

```java
IpNetwork network = (IpNetwork) localDevice.getNetwork();
InetSocketAddress inetSocketAddress = new InetSocketAddress(InetAddress.getByName(server.getHostname()), server.getPort());
network.registerAsForeignDevice(inetSocketAddress, (int) timeToLive.getDuration().getSeconds());
```

I want to handle the scenario where the device is already registered in the same way as when registration succeeds. 

Specifically:
- If the registration succeeds: great!
- If the registration fails because the device is already registered, I want the code to proceed as if the registration succeeded.
- if registration fails due to other reasons (e.g., timeout), I want to handle that separately.

I'm trying to manage the case where my device is already registered: 
I want that my code act as same if the device is register as if registration worked:
- If registration success, well.
- If registration failed but the device is already registered, well tho.

The problem is that `registerAsForeignDevice` throws a `BACnetException` in all failure cases, with the only distinguishing factor being the error message. This makes it hard to differentiate between these scenarios programmatically.

But if registration fail, (for already registered either other exception like timeout ) it return a `BACnetException` with the only difference is the error message.

## My Current Workaround

As a workaround, I used reflection to access the normally private `foreignBBMD` field in `IpNetwork` to check if the device was already registered. Here’s the workaround code:

```java
    private boolean wasAlreadyRegistered(final IpNetwork ipNetwork) {

        try {
            Field field = IpNetwork.class.getDeclaredField("foreignBBMD");
            field.setAccessible(true);
            InetSocketAddress foreignBBMD = (InetSocketAddress) field.get(ipNetwork);
            return foreignBBMD != null;

        } catch (Throwable t) {
            ...
        }
    }
```

This is, of course, not ideal because if field is renamed, code will break and there is no synchronization.

## Proposed Improvements

It would be helpful if IpNetwork provided a public method to check whether the device is already registered as a foreign device. For example:

```java
public boolean isDeviceRegisteredAsForeignDevice(){
    synchronized (foreignBBMDLock) {
            return foreignBBMD != null;
   }
}
```
In case there are some technical issue with exposing the method, I suggest to add more exceptions types.  It would be much easier to handle errors programmatically if registerAsForeignDevice threw a more specific exception (or subclasses of BACnetException) to differentiate between cases like "already registered" and other failures (e.g., timeouts).

Thank you for considering this, and I’d be happy to clarify further if needed! I also can submit a PR.

kishorevenki, 2024-11-28:
Hi Thisora,

As per the BACnet Standards, sending RegistrationAsForeignDevice to the same BBMD is acceptable even before the expiry of TimeToLive timer. In that case, the existing TimeToLive will be reset and new timer would start. Hence I feel it would be better to carry out such checking in the Application Layer. 

Handling Registration Timeout error and NAK response is already handled separately.

Thisora, 2024-11-28:
Thank you for the clarification.

I wasn’t specifically looking to differentiate between handling the Registration Timeout error and the NAK response. My main concern is that I don’t want my application to behave the same way when the BACnetException comes from the library (e.g., already registered as a foreign device) as it does when the exception is related to the FDT registration itself (such as a timeout).

kishorevenki, 2024-11-28:
Registering once again to the same BBMD while the registration is already active to the same BBMD is expected behavior and hence the library will not throw error. If the library throws error, then it is not conforming to the BACnet standard.

Thisora, 2024-11-28:
> If the library throws error, then it is not conforming to the BACnet standard.

Yes i agree, and it is what is does for now.

mlohbihler, 2026-01-05:
Closing due to inactivity. Feel free to open again if this is still an issue.

closed 2026-01-05 by mlohbihler