RadixIoT/BACnet4J #110: RequestUtils.readProperties()  The method is blocked, and there are no exception messages.

zzj957091, 2025-01-10:
@Override
    public String getPointValue(RemoteDevice remoteDevice, ObjectIdentifier oid, PropertyIdentifier propertyIdentifier) throws RuntimeException {
        PropertyValues pvs;
        String value = "";
        try {
            PropertyReferences refs = new PropertyReferences();
            refs.add(oid, propertyIdentifier);
            **pvs = RequestUtils.readProperties(localDevice, remoteDevice, refs, false, null);**
            log.info("pvs获取成功");
            Encodable result = pvs.get(oid, propertyIdentifier);
            value = result.toString();
            log.info("正常获取设备点位数据：" + value);
        } catch (Exception e) {
            log.error("获取点位值出现异常",e);
            if (propertyIdentifier.equals(PropertyIdentifier.presentValue)) {
                log.error("{} {} {}: 无法获取到 {} 属性",
                        remoteDevice.getInstanceNumber(),
                        oid.getObjectType().toString(),
                        oid.getInstanceNumber(),
                        propertyIdentifier,
                        e);
                throw new RuntimeException(remoteDevice.getInstanceNumber() + " " + oid.getObjectType().toString() + " " + oid.getInstanceNumber() + ":无法获取到" + propertyIdentifier.toString() + "属性",e);
            }
          }
        return value;
    }
    
    **pvs = RequestUtils.readProperties(localDevice, remoteDevice, refs, false, null);** is no result 。The method is blocked,It prevents my program from continuing execution

devaskim, 2025-01-14:
I suggested to add read timeout in https://github.com/MangoAutomation/BACnet4J/pull/89
Unfortunately year passed, but no feedback

kishorevenki, 2025-01-14:
Timeout is already existing in the form of PropertyIdentifier.APDUTimeOut. 
Without setting the PropertyIdentifier.SegmentationSupported and ProeprtyIdentifier.APDUTimeout of the remote device, readProperty or equivalent message cannot be sent.